pentestfunctions/BlueDucky #4: Using External BT hci1

Kalisabbb, 2024-02-16:
Hello nice work!!
I can't use with external BT adapter (hci1)  only works for hci0?
Saludos

pentestfunctions, 2024-02-16:
Yes apologies just replace hci0 in the blueducky.py script on lines 599, 602 and 624 and it will work for you :)

Kalisabbb, 2024-02-18:
Thank you I will try!

Kalisabbb, 2024-02-18:
Thank you I change I now ..no errors but can't find devices...make scan but no find..
With hcitool scan for example ok...
It's for Kali Linux...

I'm using in android with NH..and amazing!!and fun! Thanx

pentestfunctions, 2024-02-19:
Make sure you are in the bluetooth screen of the target device or have visibility on. Once you have the mac address, they don't need to be on that screen but sometimes they will need to be for it to be visible.

Kalisabbb, 2024-02-19:
Thank you for you answer...yes I try in all modes...and not find any devices with hci1

Thanx!!

closed 2024-03-29 by pentestfunctions