pentestfunctions/BlueDucky #5: No connection

ampf27, 2024-02-20:
Im able to detected bluetooth devices on raspbian

but still getting everytime

```
ERROR - ERROR connecting on port 1: [Errno 103] Software caused connection abort
ERROR - Connection failure: Connection failure on port 1
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (103, 'Software caused connection abort')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/BlueDucky/BlueDucky.py", line 264, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 103] Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/BlueDucky/BlueDucky.py", line 668, in <module>
    main()
  File "/root/BlueDucky/BlueDucky.py", line 650, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/BlueDucky/BlueDucky.py", line 625, in setup_and_connect
    establish_connections(connection_manager)
  File "/root/BlueDucky/BlueDucky.py", line 617, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/BlueDucky/BlueDucky.py", line 157, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/BlueDucky/BlueDucky.py", line 157, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/root/BlueDucky/BlueDucky.py", line 272, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 1
```

on my mobile phone I briefly get a popup witrh a pairing code but endup with the above error

What may be causing this?

Thank you

samu-delucas, 2024-02-25:
I'm getting the same error on Kali 6.6.9. Unpairing the device before running the script makes it work, but I have to do it each time I want to execute it.

ampf27, 2024-02-25:
> I'm getting the same error on Kali 6.6.9. Unpairing the device before running the script makes it work, but I have to do it each time I want to execute it.

Thank you very much for your reply.

Can you tell me how do you do that? What commands do you use?

samu-delucas, 2024-02-26:
I saw the device paired in Kali's Bluetooth GUI, so I unpaired it using that. If you didn't pair the device, then I'm not sure what you should do.

ampf27, 2024-02-26:
I have just seen the device paired with the bluetooth manager for a few seconds but then I got

```
disconnecting Bluetooth HID client
Closing L2CAPClient connection
Closing L2CAPClient connection
Closing L2CAPClient connection 
```

And disappeared from the bluetooth manager devices list. I then tried again but still not be able to run on the device

pentestfunctions, 2024-04-11:
> I'm getting the same error on Kali 6.6.9. Unpairing the device before running the script makes it work, but I have to do it each time I want to execute it.

This is interesting, so it wasn't unpairing correctly?