pentestfunctions/BlueDucky #17: tool not working on kali linux

Royalboy2000, 2024-03-28:
i followed all the steps but doesnt work 

Traceback (most recent call last):
  File "/home/samir/BlueDucky/BlueDucky.py", line 1, in <module>
    import binascii, bluetooth, sys, time, datetime, logging
ModuleNotFoundError: No module named 'bluetooth'

Royalboy2000, 2024-03-28:
2024-03-28 14:49:23,295 - ERROR - Failed to enable SSP: Can't set Simple Pairing mode on hci0: Input/output error (5)

2024-03-28 14:49:23,298 - ERROR - Error enabling SSP: Failed to enable SSP
Traceback (most recent call last):
  File "/home/samir/BlueDucky/BlueDucky.py", line 668, in <module>
    main()
  File "/home/samir/BlueDucky/BlueDucky.py", line 642, in main
    adapter.enable_ssp()
  File "/home/samir/BlueDucky/BlueDucky.py", line 111, in enable_ssp
    raise ConnectionFailureException("Failed to enable SSP")
ConnectionFailureException: Failed to enable SSP

pentestfunctions, 2024-03-29:
So the first error is possibly due to not following the full installation instructions listed in the README.md file (on the home page of this repository) while the others sems to be more in relation to possibly not being able to find the bluetooth adapter, possibly due to trying in a virtual machine? I'm not 100% sure i'd have to see your configuration. 

By default the adapter will be hci0 which is the first placement for the adapter/device. If you are in a virtual machine, you will need to use an external dongle for this to register, otherwise try and live boot.

The other issue that can arise if you are using 2 adapters is the one you might be wanting to work with will be hci1, which means going and editing the main .py file for all instances of hci0 to hci1.

baraaelgamal, 2024-03-29:
For the second error just change line 107 in BlueDucky.py to 
```python
            ssp_command = ["sudo", "hciconfig", self.iface, "sspmode"]
```

7h30th3r0n3, 2024-06-21:
Confirmed working, thx dude !!!

jarvis8669, 2025-04-15:
> For the second error just change line 107 in BlueDucky.py to
> 
>             ssp_command = ["sudo", "hciconfig", self.iface, "sspmode"]
> instead of
> 
>             ssp_command = ["sudo", "hciconfig", self.iface, "sspmode","1"]
> that "1" makes ALL the difference

its not working for me the same output is appearing