pentestfunctions/BlueDucky #18: encountered an error

BG7UHS, 2024-03-29:
[image]

pentestfunctions, 2024-03-29:
That is due to not downloading all the code. You will need the utils folder that contains the menu_functions.py file.

I'm assuming you are new to python but essentially module not found errors (as you can see on the  last line of your image) is a result of a module or package containing other functions not being installed or accessible.
In this case it can't find the utils folder then the menu_functions file. Which means it cannot load the import functions such as main_menu, read_duckyscript, run, restart_bluetooth_daemon and get_target_address - which are all functions I created for the code to work.

BG7UHS, 2024-03-29:
I used git clone, the files are complete.
[image]

pentestfunctions, 2024-03-29:
You are indeed correct, I can see the utils directory. It might best to make a file in the top level folder called __init__.py just with touch as it can be empty. Just so that other paths you might have don't interfere.

```
touch __init__.py
```

BG7UHS, 2024-03-30:
Ok 
Can be used normally

closed 2024-03-30 by BG7UHS