pentestfunctions/BlueDucky #20: Problem with launching the script which crashes after selecting a target

thepoulet153, 2024-03-30:
[image]

Hello everyone, I have a problem with launching the script which crashes after selecting a target.
I'm on: Linux kali-raspberry-pi 5.15.44-Re4son-v8l+ #1 SMP PREEMPT Debian kali-pi (2022-07-03) aarch64 GNU/Linux
Release: 2024.1
Thanks in advance

PhucHauDeveloper, 2024-03-30:
Try rebooting if you just installed it

thepoulet153, 2024-03-30:
It doesn't work, still the same error :[

thepoulet153, 2024-03-30:
It's okay, I solved my problem with my little hands and a few python magic tricks, the problem comes from the fact that the Bluetooth takes too long to restart, so I added a delay ;)

closed 2024-03-30 by thepoulet153