pentestfunctions/BlueDucky #22: Connection failure: Connection failure on port 17

HyenaTube, 2024-04-01:
im having this weird issue when trying to execute on a device, im using a raspberry pi zero 2w:

Known devices:
1: Device Name: Hyena's S21 Ultra, Address: XX:XX:XX:XX:XX:XX

Do you want to use one of these known devices? (yes/no): yes
Enter the number of the device: 1
2024-04-01 10:59:00,440 - INFO - executing 'sudo service bluetooth restart'
2024-04-01 10:59:01,478 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2024-04-01 10:59:01,526 - INFO - executing 'hciconfig hci0 name'
2024-04-01 10:59:01,532 - INFO - executing 'sudo hciconfig hci0 class 9536'
2024-04-01 10:59:01,578 - INFO - executing 'hciconfig hci0 class'
2024-04-01 10:59:01,588 - INFO - executing 'sudo hciconfig hci0 sspmode 1'
2024-04-01 10:59:03,994 - INFO - connecting to XX:XX:XX:XX:XX:XX on port 1
2024-04-01 10:59:05,294 - INFO - connecting to XX:XX:XX:XX:XX:XX on port 17
2024-04-01 10:59:05,908 - ERROR - ERROR connecting on port 17: [Errno 103] Software caused connection abort
2024-04-01 10:59:05,908 - ERROR - Connection failure: Connection failure on port 17
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (103, 'Software caused connection abort')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line Select a device by number: 1
2024-04-01 11:22:54,927 - INFO - executing 'sudo service bluetooth restart'
2024-04-01 11:22:55,965 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2024-04-01 11:22:56,011 - INFO - executing 'hciconfig hci0 name'
2024-04-01 11:22:56,017 - INFO - executing 'sudo hciconfig hci0 class 9536'
2024-04-01 11:22:56,062 - INFO - executing 'hciconfig hci0 class'
2024-04-01 11:22:56,071 - INFO - executing 'sudo hciconfig hci0 sspmode 1'
2024-04-01 11:22:58,482 - INFO - connecting to XX:XX:XX:XX:XX:XX on port 1
2024-04-01 11:22:59,606 - INFO - connecting to XX:XX:XX:XX:XX:XX on port 17
2024-04-01 11:23:00,408 - ERROR - ERROR connecting on port 17: [Errno 103] Software caused connection abort
2024-04-01 11:23:00,409 - ERROR - Connection failure: Connection failure on port 17
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (103, 'Software caused connection abort')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line 264, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 103] Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line 673, in <module>
    main()
  File "/home/kali/BlueDucky/BlueDucky.py", line 655, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 625, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/kali/BlueDucky/BlueDucky.py", line 617, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 157, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 157, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 272, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 17, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 103] Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line 673, in <module>
    main()
  File "/home/kali/BlueDucky/BlueDucky.py", line 655, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 625, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/kali/BlueDucky/BlueDucky.py", line 617, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 157, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 157, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 272, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 17

PrinzJuliano, 2024-04-03:
Same issue, different Port (1)

johseffer, 2024-04-03:
Same here, RPI4B

mauritsprogrameertnu, 2024-04-04:
I got the same. i use a rpi4 running raspberry pi os lite 64bit

AyoubMrani, 2024-04-16:
`apt-get update
apt-get install python3 python3-pip python3-dev git libssl-dev libffi-dev build-essential
python3 -m pip install --upgrade pip
python3 -m pip install --upgrade pwntools`

do this and It will work

daframe2r, 2024-05-06:
You can try to add sleep time after connection (worked for me): 
connection_manager.create_connection(1)   # SDP
time.sleep(1)
connection_manager.create_connection(17)  # HID Control
time.sleep(1)

mohamedmajid91, 2024-05-19:
same error again

> You can try to add sleep time after enable ssp (worked for me)
> 
> line 671: adapter.enable_ssp() time.sleep(2)

same error again

HimanshuAttri, 2024-10-20:
#below
adapter.enable_ssp()
#add
time.sleep(5)

Worked for me on RPi