pentestfunctions/BlueDucky #23: Failed to enable SSP

spideyshanexample, 2024-04-01:
2024-04-01 22:10:27,385 - ERROR - Failed to enable SSP: Can't set Simple Pairing mode on hci0: Input/output error (5)                                     
                                                                             
2024-04-01 22:10:27,386 - ERROR - Error enabling SSP: Failed to enable SSP
Traceback (most recent call last):
  File "/root/Desktop/BlueDucky/BlueDucky.py", line 673, in <module>
    main()
  File "/root/Desktop/BlueDucky/BlueDucky.py", line 647, in main
    adapter.enable_ssp()
  File "/root/Desktop/BlueDucky/BlueDucky.py", line 111, in enable_ssp
    raise ConnectionFailureException("Failed to enable SSP")
ConnectionFailureException: Failed to enable SSP

``
**For the first time it run perfectly. When I try to run again it show error . 
I am using Kali Linux**

Vuxorr, 2024-05-12:
change in scriot line from :

 ssp_command = ["sudo", "hciconfig", self.iface, "sspmode", "1"]

to

 ssp_command = ["sudo", "hciconfig", self.iface, "sspmode"]

without 1 at the end. I work for me.

6nsm, 2024-05-14:
[BlueDucky.txt](https://github.com/pentestfunctions/BlueDucky/files/15309986/BlueDucky.txt)
USE THIS COPY IT AND PASTE THIS TO YOUR ACTUAL BLUEDUCKY.PY

Error13784, 2024-06-29:
> [BlueDucky.txt](https://github.com/pentestfunctions/BlueDucky/files/15309986/BlueDucky.txt) USE THIS COPY IT AND PASTE THIS TO YOUR ACTUAL BLUEDUCKY.PY

works! thank you!

hadelbns, 2024-09-28:
thank u

Ryzzen, 2025-01-06:
> change in scriot line from :
> 
> ssp_command = ["sudo", "hciconfig", self.iface, "sspmode", "1"]
> 
> to
> 
> ssp_command = ["sudo", "hciconfig", self.iface, "sspmode"]
> 
> without 1 at the end. I work for me.

This does not set the pairing mode, all it does is read the currently set mode.

Cf hciconfig source code:

```c
if (opt) {
		uint8_t mode = atoi(opt);

		if (hci_write_simple_pairing_mode(dd, mode, 2000) < 0) {
			fprintf(stderr, "Can't set Simple Pairing mode on hci%d: %s (%d)\n",
					hdev, strerror(errno), errno);
			exit(1);
		}
	} else {
		uint8_t mode;

		if (hci_read_simple_pairing_mode(dd, &mode, 1000) < 0) {
			fprintf(stderr, "Can't read Simple Pairing mode on hci%d: %s (%d)\n",
					hdev, strerror(errno), errno);
			exit(1);
		}
```

jarvis8669, 2025-04-15:
hey i am using kali linux and i have same problem 
is there anybody can sortout it