pentestfunctions/BlueDucky #24: Devices asking to be paired

beraoudabdelkhalek, 2024-04-01:
I need to pair the device manually to send the keystrokes, I have a Qualcomm Atheros qca9377 Bluetooth driver.

beraoudabdelkhalek, 2024-04-08:
No, I attacked a Huawei Nova 3i (Android 9), a OnePlus Nord 100 (Android 11), and a Xiaomi phone with Android 10. This is why I thought the problem was in my Bluetooth card.

crackedpotato007, 2024-05-16:
Same issue with Samsung A10, Android 12 using Intel AX201

Enveeus, 2024-05-21:
Same for Samsung S10 using Android 12, But if you refer to the CVE, Andriod 12 is not on the list.. It skips 12 and moves to 13, 14. So I'm taking that when patched, the device will request pairing.

Enveeus, 2024-05-25:
So this is an odd one. I ran this on an  Samsung A10 with Android 9 and the first time around the script made it all the way through but failed to load the web page I defined in payload 1.. which was YouTube. So I ran two more attempts using the same payload with no success.. the web page failed to open on the phone. On the 4th and every time there after  I get the prompt to pair on the phone now..