pentestfunctions/BlueDucky #25: Not running on my Kali(2024.1), different kind of error compared to the other ones

Ninjaex13, 2024-04-04:
Hey, there I've literally installed everything step by step from the readme.md file but everytime when I try to run the script even with sudo enabled I get the following error(my bluetooth card(hci0) is perfectly functional and I have no idea why im getting the following error)), I'd be more than thankful if someone could help me out <3

2024-04-04 15:27:57,634 - INFO - executing 'sudo service bluetooth restart'
2024-04-04 15:27:58,496 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
Traceback (most recent call last):
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 673, in <module>
    main()
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 646, in main
    adapter = setup_bluetooth(target_address, adapter_id)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 603, in setup_bluetooth
    adapter.set_property("name", "Robot")
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 87, in set_property
    self._run_command(command)
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 81, in _run_command
    raise ConnectionFailureException(f"Failed to execute command: {' '.join(command)}. Error: {result.stderr}")
ConnectionFailureException: Failed to execute command: sudo hciconfig hci0 name Robot. Error: Can't change local name on hci0: Network is down (100)

Ninjaex13, 2024-04-05:
Select a device by number: 2
2024-04-05 18:30:07,066 - INFO - executing 'sudo service bluetooth restart'
2024-04-05 18:30:07,885 - INFO - executing 'sudo hciconfig hci0 name Andrew'
Traceback (most recent call last):
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 673, in <module>
    main()
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 646, in main
    adapter = setup_bluetooth(target_address, adapter_id)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 603, in setup_bluetooth
    adapter.set_property("name", "Andrew")
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 87, in set_property
    self._run_command(command)
  File "/home/andrew/Downloads/BlueDucky/BlueDucky.py", line 81, in _run_command
    raise ConnectionFailureException(f"Failed to execute command: {' '.join(command)}. Error: {result.stderr}")
ConnectionFailureException: Failed to execute command: sudo hciconfig hci0 name Andrew. Error: Can't change local name on hci0: Network is down (100)

CarrotRub, 2024-04-07:
It is probably because your bluetooth device is not named hci0 or that you possibly don't even have a bluetooth adapter, try by using `sudo hciconfig` and send me what you found. (Hide your BD Address)

GasStationWeenie, 2024-04-08:
In /utils/menu_functions.py you will need to change the time.sleep value under "def restart_bluetooth_daemon();" to something higher than .5.  I went with 5 and it's fine now.

Stallion77RepoOfficial, 2024-12-12:
> /utils/menu_functions.py'de "def restart_bluetooth_daemon();" altındaki time.sleep değerini .5'ten daha yüksek bir değere değiştirmeniz gerekecek. Ben 5'i kullandım ve şimdi iyi durumda.

scan option still not working but working with manual