pentestfunctions/BlueDucky #29: ERROR connecting on port 17: [Errno 103] Software caused connection abort

SmartBoyMuzaffar, 2024-04-22:
Enter the number of the payload you want to load: 1
Selected payload: /home/smartboy/BlueDucky/payloads/payload_example_1.txt
2024-04-23 04:05:34,694 - INFO - executing 'sudo service bluetooth restart'
2024-04-23 04:05:35,873 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2024-04-23 04:05:35,943 - INFO - executing 'hciconfig hci0 name'
2024-04-23 04:05:35,952 - INFO - executing 'sudo hciconfig hci0 class 9536'
2024-04-23 04:05:36,005 - INFO - executing 'hciconfig hci0 class'
2024-04-23 04:05:36,019 - INFO - executing 'sudo hciconfig hci0 sspmode 1'
2024-04-23 04:05:40,290 - INFO - connecting to 0C:9A:3C:41:96:5E on port 1
2024-04-23 04:05:41,045 - INFO - connecting to 0C:9A:3C:41:96:5E on port 17
2024-04-23 04:05:41,917 - ERROR - ERROR connecting on port 17: [Errno 103] Software caused connection abort
2024-04-23 04:05:41,918 - ERROR - Connection failure: Connection failure on port 17
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (103, 'Software caused connection abort')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 265, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 103] Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 697, in <module>
    main()
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 679, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 626, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 618, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 158, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 158, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/smartboy/BlueDucky/BlueDucky.py", line 273, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 17

hexdrx, 2024-05-12:
do u solve it?

SmartBoyMuzaffar, 2024-05-12:
> do u solve it?
> 
> 

no

mohamedmajid91, 2024-05-15:
i have the same error

hexdrx, 2024-05-17:
Add time.sleep(3). That works for me

line 671:
adapter.enable_ssp()
time.sleep(3)

Neowizard, 2024-06-03:
Had the same issue as all the others in this thread, the workaround proposed by @hexdrx worked perfectly.

On thing, the line number must be based on an older commit, since it's not where `adapter.enable_ssp()` is located now. Regardless, adding `time.sleep(3)` after the call to `enable_ssp` works