pentestfunctions/BlueDucky #30: How to fix this?

RwOnke, 2024-04-24:
python3 BlueDucky.py
Traceback (most recent call last):
  File "/root/BlueDucky/BlueDucky.py", line 7, in <module>
    from utils.menu_functions import (main_menu, read_duckyscript, run, restart_bluetooth_daemon, get_target_address)
ModuleNotFoundError: No module named 'utils.menu_functions'; 'utils' is not a package

SonofShiva, 2024-05-13:
Same problem

Feifel81, 2024-05-13:
The files are in the wrong directory:
1: go to `~/pybluez/BlueDucky/utils`
2:Copy the Files in the right folder `sudo cp magic_keyboard_hid.py menu_functions.py _pycache_ register_device.py ~/pybluez/BlueDucky`
3: go to `~/pybluez/BlueDucky`
4: Edit the BlueDucky.py with:`sudo mousepad BlueDucky.py`
5: Edit line 7 & 8 remove the "utils." in each line and **SAVE the File**
It should look like this :
Line 7:`from menu_functions import (main_menu, read_duckyscript, run, restart_bluetooth_daemon, get_target_address)
Line 8: from register_device import register_hid_profile, agent_loop`
[image]
6: It should work now ........

SonofShiva, 2024-05-15:
Hey that worked, thank you very much

RwOnke, 2024-06-05:
> @SonofShiva @RwOnke The files are in the wrong directory: 1: go to `~/pybluez/BlueDucky/utils` 2:Copy the Files in the right folder `sudo cp magic_keyboard_hid.py menu_functions.py _pycache_ register_device.py ~/pybluez/BlueDucky` 3: go to `~/pybluez/BlueDucky` 4: Edit the BlueDucky.py with:`sudo mousepad BlueDucky.py` 5: Edit line 7 & 8 remove the "utils." in each line and **SAVE the File** It should look like this : Line 7:`from menu_functions import (main_menu, read_duckyscript, run, restart_bluetooth_daemon, get_target_address) Line 8: from register_device import register_hid_profile, agent_loop` ![Mousepad BlueDucky](https://private-user-images.githubusercontent.com/169727971/330226904-31e473fa-bad0-4da3-9395-12dcaecd311f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1NTU5NTYsIm5iZiI6MTcxNzU1NTY1NiwicGF0aCI6Ii8xNjk3Mjc5NzEvMzMwMjI2OTA0LTMxZTQ3M2ZhLWJhZDAtNGRhMy05Mzk1LTEyZGNhZWNkMzExZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNVQwMjQ3MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZDcyZGZhN2ViYTdhOTc0MTAyOWMwOGI1YzViZGUyNDQxMmU4ZGFhNDIwMzI1NzNmNTQxNDVhMzFkZTNhMDE5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-Qdqi8avuQTwauP6ElyeGH6E1keJ8wjbIp5aZPbOOmI) 6: It should work now ........
> 
> fixes #30 @pentestfunctions [Merge pull request](https://github.com/pentestfunctions/BlueDucky/issues/30) This is [completed](https://github.com/pentestfunctions/BlueDucky/issues/30)

Hey there, This worked well for me but after reboot it seems to have broken my internal bluetooth adapter, my bluetooth mouse doesnt pair and blueducky doesn't find any bluetooth devices, hcitool dev shows no devices either. Any idea on how to fix this??

Feifel81, 2024-06-05:
@RwOnke Check with `journalctl | grep "bluetooth"` for error messages related to bluetooth. In general you can check with journalctl in Terminal for Errors and get a hint what to look for. Difficult to say. 
Try running these in terminal to see if it will wake your bluetooth up:
`sudo systemctl enable bluetooth.service`
`sudo systemctl start bluetooth.service`