pentestfunctions/BlueDucky #31: Failed to execute command: sudo hciconfig hci0 name Robot POC.

omgaikwad3001, 2024-04-27:
Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line 697, in <module>
    main()
  File "/home/kali/BlueDucky/BlueDucky.py", line 670, in main
    adapter = setup_bluetooth(target_address, adapter_id)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 604, in setup_bluetooth
    adapter.set_property("name", "Robot POC")
  File "/home/kali/BlueDucky/BlueDucky.py", line 88, in set_property
    self._run_command(command)
  File "/home/kali/BlueDucky/BlueDucky.py", line 82, in _run_command
    raise ConnectionFailureException(f"Failed to execute command: {' '.join(command)}. Error: {result.stderr}")
ConnectionFailureException: Failed to execute command: sudo hciconfig hci0 name Robot POC. Error: Can't change local name on hci0: Network is down (100)

daframe2r, 2024-05-06:
You need to had sleep time because the bluetooth is probably not restarted yet.

4m3rr0r, 2024-05-12:
If you're encountering ongoing issues, you might want to consider making the following modification to the code

[image]

[image]