pentestfunctions/BlueDucky #36: No module named 'bluetooth'

djaberoff, 2024-05-13:
[image]

what should i do ?

djaberoff, 2024-05-13:
oh, i remembered something to ask, do i need linux to run it or i can run it on windows

abdulhakkeempa, 2024-05-13:
@djaberoff I tried it on Linux. Faces the same issues, so it is not a platform dependent issue.

djaberoff, 2024-05-13:
> @djaberoff I tried it on Linux. Faces the same issues, so it is not a platform dependent issue.

oh thank you i was going to switch to linux, u really saved me <3

abdulhakkeempa, 2024-05-13:
> > @djaberoff I tried it on Linux. Faces the same issues, so it is not a platform dependent issue.
> 
> oh thank you i was going to switch to linux, u really saved me <3

Oh, found that the issue was caused due to my Conda Environment. When I disabled it, it worked well. I advise you to try it own Linux environment itself.

pragnesh-singh-rajput, 2024-05-13:
I got the same error at first but when i did the setup properly it worked

Do this setup first then run the py file 

# update apt
sudo apt-get update
sudo apt-get -y upgrade

# install dependencies from apt
sudo apt install -y bluez-tools bluez-hcidump libbluetooth-dev \
                    git gcc python3-pip python3-setuptools \
                    python3-pydbus

# install pybluez from source
git clone https://github.com/pybluez/pybluez.git
cd pybluez
sudo python3 setup.py install

# build bdaddr from the bluez source
cd ~/
git clone --depth=1 https://github.com/bluez/bluez.git
gcc -o bdaddr ~/bluez/tools/bdaddr.c ~/bluez/src/oui.c -I ~/bluez -lbluetooth
sudo cp bdaddr /usr/local/bin/

djaberoff, 2024-05-13:
> sudo apt-get update

how do i do it please <<<3

djaberoff, 2024-05-13:
> I got the same error at first but when i did the setup properly it worked
>
> Do this setup first then run the py file
>
> # update apt
> sudo apt-get update sudo apt-get -y upgrade
>
> # install dependencies [...]

it sais that sudo is not recognised

angel6uard, 2024-05-14:
you're on windows lol

djaberoff, 2024-05-14:
> you're on windows lol

do i need linux ?

angel6uard, 2024-05-14:
> > you're on windows lol
> 
> do i need linux ?

yes.

djaberoff, 2024-05-14:
> > > you're on windows lol
> > 
> > 
> > do i need linux ?
> 
> yes.

ooh tysm <<<<<3333