pentestfunctions/BlueDucky #39: isue on kali linux failed enabling SSP

6nsm, 2024-05-14:
[BlueDucky.txt](https://github.com/pentestfunctions/BlueDucky/files/15309930/BlueDucky.txt)
  IF YOU FACE THIS PROBLEM CONSIDER USING FILE ABOVE

Chipeza, 2024-05-15:
[image]

Chipeza, 2024-05-15:
It's not fixing the problem

Chipeza, 2024-05-15:
How can i use that text you have send

ziriolo, 2024-05-15:
Enter the number of the payload you want to load: 1
Selected payload: /home/user/BlueDucky/payloads/payload_example_2.txt
2024-05-15 19:40:28,083 - INFO - executing 'sudo service bluetooth restart'
2024-05-15 19:40:28,857 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2024-05-15 19:40:28,936 - INFO - executing 'hciconfig hci0 name'
2024-05-15 19:40:28,959 - INFO - executing 'sudo hciconfig hci0 class 9536'
2024-05-15 19:40:29,008 - INFO - executing 'hciconfig hci0 class'
2024-05-15 19:40:29,021 - INFO - executing 'sudo hciconfig hci0 sspmode'
2024-05-15 19:40:29,059 - ERROR - Failed to enable SSP: Can't read Simple Pairing mode on hci0: Input/output error (5)
                                                                                                                      
2024-05-15 19:40:29,060 - ERROR - Error enabling SSP: Failed to enable SSP
Traceback (most recent call last):
  File "/home/user/BlueDucky/BlueDucky.py", line 697, in <module>
    main()
  File "/home/user/BlueDucky/BlueDucky.py", line 671, in main
    adapter.enable_ssp()
  File "/home/user/BlueDucky/BlueDucky.py", line 112, in enable_ssp
    raise ConnectionFailureException("Failed to enable SSP")
ConnectionFailureException: Failed to enable SSP

ziriolo, 2024-05-15:
Selected payload: /home/user/BlueDucky/payloads/payload_example_2.txt
2024-05-15 20:01:47,347 - INFO - executing 'sudo service bluetooth restart'
2024-05-15 20:01:47,992 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2024-05-15 20:01:48,032 - INFO - executing 'hciconfig hci0 name'
2024-05-15 20:01:48,051 - ERROR - Unable to set adapter name, aborting. Output: hci0:   Type: Primary  Bus: USB
        BD Address: 00:1A:7D:DA:71:13  ACL MTU: 679:8  SCO MTU: 48:16                                                 
        Name: 'CSR8510 A10.'                                                                                          
                                                                                                                      
Traceback (most recent call last):
  File "/home/user/BlueDucky/BlueDucky.py", line 697, in <module>
    main()
  File "/home/user/BlueDucky/BlueDucky.py", line 670, in main
    adapter = setup_bluetooth(target_address, adapter_id)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BlueDucky/BlueDucky.py", line 604, in setup_bluetooth
    adapter.set_property("name", "Robot POC")
  File "/home/user/BlueDucky/BlueDucky.py", line 95, in set_property
    raise ConnectionFailureException(f"Failed to set {prop}")
ConnectionFailureException: Failed to set name

6nsm, 2024-05-16:
> It's not fixing the problem

Just download it by the touching on it

Lamentomori, 2024-05-19:
> [BlueDucky.txt](https://github.com/pentestfunctions/BlueDucky/files/15309930/BlueDucky.txt) IF YOU FACE THIS PROBLEM CONSIDER USING FILE ABOVE

THERE IS NO NEED FOR THIS. IF YOU RUN DUCKYSCRIPT.PY AS SUDO

6nsm, 2024-05-19:
Dude if you think it's not working message me at 
www.instagram.com/ekoahamdutivnasti

Roli810, 2024-05-25:
so did that file fix your issue? I still have the same error

Roli810, 2024-05-25:
I was using a bluetooth 2.0 EDR (which is basically bluetooth 2.1) adapter, and the pairing works a little bit differently. I am trying to find a workaround but if you type 0000 and 0000 on pc too in order to pair, the exploit works. If i find a workaround, I'll send the code

7h30th3r0n3, 2024-06-21:
> [BlueDucky.txt](https://github.com/pentestfunctions/BlueDucky/files/15309930/BlueDucky.txt) IF YOU FACE THIS PROBLEM CONSIDER USING FILE ABOVE

maybe explain what you do ?

jarvis8669, 2025-04-15:
hey i am using an laptop with kali-linux and i have same problem i there anybody can  sortout it 
"Selected payload: /home/jarvis/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky/payloads/wp_payload.txt
2025-04-15 19:44:37,578 - INFO - executing 'sudo service bluetooth restart'
2025-04-15 19:44:38,735 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2025-04-15 19:44:38,844 - INFO - executing 'hciconfig hci0 name'
2025-04-15 19:44:38,860 - INFO - executing 'sudo hciconfig hci0 class 9536'
2025-04-15 19:44:38,979 - INFO - executing 'hciconfig hci0 class'
2025-04-15 19:44:39,003 - INFO - executing 'sudo hciconfig hci0 sspmode 1'
2025-04-15 19:44:39,114 - ERROR - Failed to enable SSP: Can't set Simple Pairing mode on hci0: Input/output error (5)

2025-04-15 19:44:39,118 - ERROR - Error enabling SSP: Failed to enable SSP
Traceback (most recent call last):
  File "/home/jarvis/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky.py", line 755, in <module>
    main()
    ~~~~^^
  File "/home/jarvis/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky.py", line 720, in main
    adapter.enable_ssp()
    ~~~~~~~~~~~~~~~~~~^^
  File "/home/jarvis/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky/BlueDucky.py", line 111, in enable_ssp
    raise ConnectionFailureException("Failed to enable SSP")
ConnectionFailureException: Failed to enable SSP
"
this is the output when i run the program 
(i teted the solution remove 1 from the line 107 )

ekoahamdutivnasti, 2025-04-15:
Just use my blueducky.txt replace it with your file you good to go

jarvis8669, 2025-04-15:
> Just use my blueducky.txt replace it with your file you good to go

i replaced that but not working