pentestfunctions/BlueDucky #41: OSError: [Errno 19] No such device. how to fix it

djaberoff, 2024-05-15:
Attempting to scan now...
Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky/BlueDucky.py", line 697, in <module>
    main()
  File "/home/kali/BlueDucky/BlueDucky/BlueDucky.py", line 637, in main
    target_address = get_target_address()
                     ^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky/utils/menu_functions.py", line 7, in get_target_address
    devices = scan_for_devices()
              ^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky/utils/menu_functions.py", line 126, in scan_for_devices
    nearby_devices = bluetooth.discover_devices(duration=8, lookup_names=True, flush_cache=True, lookup_class=True)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/PyBluez-0.30-py3.11-linux-x86_64.egg/bluetooth/bluez.py", line 19, in discover_devices
OSError: [Errno 19] No such device

please help me

xensurve, 2024-05-15:
systemctl is-enabled bluetooth 

hcitool device

coldvisionz, 2024-05-16:
> systemctl is-active bluetooth # check if bluetooth is running
> 
> hcitool device # lists your devices

also, bluetooth is disabled by default on kali distros

`sudo systemctl enable bluetooth
sudo systemctl start bluetooth`

djaberoff, 2024-05-18:
> > systemctl is-active bluetooth # check if bluetooth is running
> > hcitool device # lists your devices
> 
> also, bluetooth is disabled by default on kali distros
> 
> `sudo systemctl enable bluetooth`
> 
> `sudo systemctl start bluetooth`

i turned it on and the same problem

djaberoff, 2024-05-18:
> systemctl is-active bluetooth # check if bluetooth is running
> 
> hcitool device # lists your devices

i checked it and it said thats inactive and i turned it on then it said active but the same problem

closed 2024-05-18 by djaberoff

coldvisionz, 2024-05-18:
did you solve the problem ? 
can you share the whole output from the terminal and logs

Alexalecos, 2025-02-08:
If you still have the problem half a year later, try sudo hciconfig hci0 up