pentestfunctions/BlueDucky #42: ModuleNotFoundError: No module named '_dbus_bindings'

krt1k, 2024-05-15:
Can't able to start Blueducky.py

`git clone https://github.com/pentestfunctions/BlueDucky.git
cd BlueDucky
sudo hciconfig hci0 up
python3 BlueDucky.py
Cloning into 'BlueDucky'...
remote: Enumerating objects: 162, done.
remote: Counting objects: 100% (60/60), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 162 (delta 35), reused 43 (delta 22), pack-reused 102
Receiving objects: 100% (162/162), 803.75 KiB | 3.32 MiB/s, done.
Resolving deltas: 100% (69/69), done.
Traceback (most recent call last):
  File "/home/rently/BlueDucky/BlueDucky.py", line 8, in <module>
    from utils.register_device import register_hid_profile, agent_loop
  File "/home/rently/BlueDucky/utils/register_device.py", line 1, in <module>
    import dbus
  File "/usr/lib/python3/dist-packages/dbus/__init__.py", line 73, in <module>
    import dbus.types as types
  File "/usr/lib/python3/dist-packages/dbus/types.py", line 8, in <module>
    from _dbus_bindings import (
ModuleNotFoundError: No module named '_dbus_bindings'
`

xensurve, 2024-05-15:
install dbus-python

krt1k, 2024-05-16:
@xensurve, I have already installed dbus-python as well as pydbus.

```
❯ python --version
Python 3.10.14
❯ pip --version
pip 24.0 from /home/rently/.local/lib/python3.10/site-packages/pip (python 3.10)
❯ pip install dbus-python
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: dbus-python in /usr/lib/python3/dist-packages (1.2.16)
❯ pip install pydbus
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pydbus in /usr/lib/python3/dist-packages (0.6.0)
```

krt1k, 2024-05-16:
Update: 

After I pulled the latest commit this morning:

```
❯ python BlueDucky.py
=============================================================================================================================================================================
                                                                    BlueDucky - Bluetooth Device Attacker                                                                    
                                                                                   Ver 2.1                                                                                   
=============================================================================================================================================================================
Remember, you can still attack devices without visibility..
If you have their MAC address..
=============================================================================================================================================================================
Traceback (most recent call last):
  File "/home/USER/BlueDucky/BlueDucky.py", line 9, in <module>
    from utils.register_device import register_hid_profile, agent_loop
  File "/home/USER/BlueDucky/utils/register_device.py", line 1, in <module>
    import dbus
  File "/usr/lib/python3/dist-packages/dbus/__init__.py", line 73, in <module>
    import dbus.types as types
  File "/usr/lib/python3/dist-packages/dbus/types.py", line 8, in <module>
    from _dbus_bindings import (
ModuleNotFoundError: No module named '_dbus_bindings'
```

Getting this error