pentestfunctions/BlueDucky #47: BlueDucky.py script run and get stuck in Kali Linux

devbiswas, 2024-05-17:
[image]

I have followed all the installation process mentioned in the READMI.md file without any error. but when I run BlueDucky.py script it appears and get stuck. How can I solve it?

MrGitHub5000, 2024-05-17:
Same issue for me, running latest Kali on Raspberry Pi 5

Roshan-777, 2024-05-17:
Same issue man, I even tried using two machines.
Still facing the same issue

Lamentomori, 2024-05-19:
> ![Screenshot_2024-05-17_13_28_49](https://private-user-images.githubusercontent.com/33261262/331505034-cb16ab84-8250-4a54-ac60-5d8bf4e1948b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzc0ODEsIm5iZiI6MTcxNjEzNzE4MSwicGF0aCI6Ii8zMzI2MTI2Mi8zMzE1MDUwMzQtY2IxNmFiODQtODI1MC00YTU0LWFjNjAtNWQ4YmY0ZTE5NDhiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE2NDYyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1YzhlNTliOGY3ZjVhNDk0ZTk2YjUxNzA4ZDlkNGZkODM3NzljMWQwZTMzMjFkMGFiMmRiMWU0MWQwZGZhNTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KRQ0PVIqYDAG1tY7vMWUUnkMa-GSe8CYmftEaUXYnWk)
> 
> I had followed all the installation process mentioned in the READMI.md file without any error. but when I run BlueDucky.py script it appears and get stuck. How can I solve it?

Hey there! Your problem is Interesting and Unique! Please contact me via Discord so I can resolve your problem: @youjsgotpwned

Roshan-777, 2024-05-20:
> 
> Hey there! Your problem is Interesting and Unique! Please contact me via Discord so I can resolve your problem: @youjsgotpwned

Can't find you on discord, is that username correct?

Roshan-777, 2024-05-21:
> Same issue for me, running latest Kali on Raspberry Pi 5

Did you get a solution to the problem?

Roshan-777, 2024-05-21:
> No, I also connected to Mac laptop with the same result within kali using VM :(

Damn, it sucks.
l just saw that someone has addressed this problem again on this forum.

kali-bard, 2024-05-27:
same problem here. cant even see the image of "peace was never an option". just some text and stuck at that point. hey @Roshan-777 if you get the fix for that, could you please put it up on Github? thanks.

kali-bard, 2024-05-27:
psst... i think this might help people: https://github.com/pentestfunctions/BlueDucky/issues/51

MrGitHub5000, 2024-05-27:
-- Much love --
Thanks for linking.
Happened to be playing with it for the first time in ages when you posted.
WORKING NOW !