pentestfunctions/BlueDucky #49: I get the error No Compatible {blue}Bluetooth devices{reset} are connected.

Aggelos11, 2024-05-18:
[image]
[image]
[image]
[image]

DG1FI, 2024-05-18:
Same - with internal RPI4 BT Device and an Cambridge Stick.

YKTV8, 2024-05-18:
Confirmed. Same issue. RPI4 Asus BT adapter.

Aggelos11, 2024-05-19:
I don’t think it’s the adaptor problem . The developer modified the script some days before and probably he broke something . He needs to fix it asap

Lamentomori, 2024-05-19:
> I don’t think it’s the adaptor problem . The developer modified the script some days before and probably he broke something . He needs to fix it asap

Its a problem with your adapter. If you check BlueDucky.py, on Line: 640 We have a new Function Called "troubleshoot_bluetooth". The purpose of this function is to troubleshoot the adapters capability with bluetoothctl, if the adapter is not capatible with Bluetoothctl, or Bluetoothctl has been installed incorrectly you will be notified of whats preventing you from accessing the application. It is designed to keep people who are not capable of accessing the application later from accessing it initially..

Please Reference the code found in this function

```
 # Check if bluetoothctl is available
    try:
        subprocess.run(['bluetoothctl', '--version'], check=True, stdout=subprocess.PIPE)
    except subprocess.CalledProcessError:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: {blue}bluetoothctl {reset}is not installed or not working properly.")
        return False

    # Check for Bluetooth adapters
    result = subprocess.run(['bluetoothctl', 'list'], capture_output=True, text=True)
    if "Controller" not in result.stdout:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: No {blue}Bluetooth adapters{reset} have been detected.")
        return False

    # List devices to see if any are connected
    result = subprocess.run(['bluetoothctl', 'devices'], capture_output=True, text=True)
    if "Device" not in result.stdout:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.")
        return False

    # if no issues are found then continue
    return True
```

Lamentomori, 2024-05-19:
If you think this may be a mistake, please contact me via discord with anything that may help your problem if it does NOT concern your Bluetooth Adapter, If your Bluetooth adapter is not supported, Then it is NOT supported. You will need to use another device or purchase a new Adapter, and I will work with you to fix the problem | I apologize for any inconvenience. @youjsgotpwned

Aggelos11, 2024-05-19:
Well my Bluetooth adaptor is Asus USB-BT500 USB Bluetooth 5.0 . Do you know if I should install any drivers or something ? Because it recognised by Kali and I can scan for Bluetooth devices using the hcitool. Do you have any adaptor to suggest ?

Lamentomori, 2024-05-19:
> ![Screenshot from 2024-05-18 10-53-19](https://private-user-images.githubusercontent.com/62383412/331810132-72ec01bd-33ac-481b-b386-1316914fa31f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzY4MjYsIm5iZiI6MTcxNjEzNjUyNiwicGF0aCI6Ii82MjM4MzQxMi8zMzE4MTAxMzItNzJlYzAxYmQtMzNhYy00ODFiLWIzODYtMTMxNjkxNGZhMzFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE2MzUyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg2NzE2ZjU1NjY0Y2FkZTIzZjlkNTUzNTY5OTU3ZTVjNzIxNzM5NDQ0N2U1MjcwNGNlM2UyMmZmYTBjYjcxY2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NVAD2nTUGpJqhFT-JwQ8ES_CRM7V5-cixMk5Fg9Be2k) ![Screenshot from 2024-05-18 10-53-57](https://private-user-images.githubusercontent.com/62383412/331810135-28271bdf-8839-48c3-a827-ecacad580bde.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzY4MjYsIm5iZiI6MTcxNjEzNjUyNiwicGF0aCI6Ii82MjM4MzQxMi8zMzE4MTAxMzUtMjgyNzFiZGYtODgzOS00OGMzLWE4MjctZWNhY2FkNTgwYmRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE2MzUyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhMTZkNWQ4OTI2NTNkODNlYzZiNTUyNWFjZjVmOTg1OWY4YmViYzMyYzdiZTA1ODhmZGQ5YzE0NmU3MTE1NWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Gj4M9TSB4L692El4T_IN51jMokAfb7amz47gyYuLXtg) ![Screenshot from 2024-05-18 10-55-13](https://private-user-images.githubusercontent.com/62383412/331810136-4f0c0b07-e4e6-4087-851d-af3d4bdc510f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzY4MjYsIm5iZiI6MTcxNjEzNjUyNiwicGF0aCI6Ii82MjM4MzQxMi8zMzE4MTAxMzYtNGYwYzBiMDctZTRlNi00MDg3LTg1MWQtYWYzZDRiZGM1MTBmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE2MzUyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlZDJhZjdlZDE1ZWQ5MjRhOWE0N2JhY2ExYTE2YmRjNzYyODE0YzI1YzNhMWQzN2NhNTRkM2VkNWVlNTFhNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-Bhv_CEfYY0OwNGHw0qUZFWmEBONfJ8J3OTpgUEKevU) ![Screenshot from 2024-05-18 10-55-41](https://private-user-images.githubusercontent.com/62383412/331810137-2009d5f7-9236-4d90-864a-7b2965052dc1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMzY4MjYsIm5iZiI6MTcxNjEzNjUyNiwicGF0aCI6Ii82MjM4MzQxMi8zMzE4MTAxMzctMjAwOWQ1ZjctOTIzNi00ZDkwLTg2NGEtN2IyOTY1MDUyZGMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDE2MzUyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkwMzhlOGUyNjg5YzIzMDFjNzVlYjY4MjcyYTI0YWZmZGQ3MGZmYWU3NDIyNTc4ZDdhNDFjNDQ4OTI3ODhkNDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1Mlr60GRZYSoACDC6FmbshkzLokba6sXi0iMQstYnyA)

I do apologise, I did not see the screenshot.

Lamentomori, 2024-05-19:
> Well my Bluetooth adaptor is Asus USB-BT500 USB Bluetooth 5.0 . Do you know if I should install any drivers or something ? Because it recognised by Kali and I can scan for Bluetooth devices using the hcitool. Do you have any adaptor to suggest ?

within BlueDucky.py the original developer Hardcoded Hci0 causing issues. I added this to make debugging easier, I will make a modification to check for another Bluetooth adapter as it defaults to hci0. For now Please open BlueDucky.py and 'Control + F' and find hci0 and replace it with hci1.

Lamentomori, 2024-05-19:
> within BlueDucky.py the original developer Hardcoded Hci0 causing issues. I added this to make debugging easier, I will make a modification to check for another Bluetooth adapter as it defaults to [...]

Doing what I described above, should resolve your issues. As right now its defaulting to an adapter that is `DOWN` or currently Not Available. Feel free to contact me via discord: @youjsgotpwned

Aggelos11, 2024-05-19:
Yes but as you can see in the screenshot I sent I already replaced hci0 with hci1 in the line 674 I think where it says default=… . Should I replace it to other points as well ?

Aggelos11, 2024-05-19:
Also can you check if my adaptor is supported pls or not ? Because as I told you it can scan for Bluetooth devices using the the hcitool which I think it’s a sign that it should work

alex8819, 2024-05-19:
i solved with this:
sudo systemctl restart bluetooth

Start the process:
bluetoothctl

execute command:
power on
agent on
default-agent
scan on
exit

after that sudo python3 BlueDucky.py will start.

Aggelos11, 2024-05-19:
Do I need to do this every time I wanna use it or only once ?

alex8819, 2024-05-19:
After BlueDucky.py crash, every time i need to make this command 2 times, after that BlueDucky.py start, but when it will crash i need to do it again. i mada a small script to speed up:
`sudo nano reset.sh`
```
#!/bin/bash
# Script per ripristinare il Bluetooth dopo un crash di BlueDucky.py

echo "Ripristino del Bluetooth in corso..."
sudo systemctl restart bluetooth
bluetoothctl power on
bluetoothctl agent on
bluetoothctl default-agent
bluetoothctl scan on
```
Add permission:
`sudo chmod +x reset.sh`

execute the script:
`./reset.sh`

Stop the script
CTRL + C

Execute agin,stop again and start:
sudo python3 BlueDucky.py

Aggelos11, 2024-05-19:
Which chipset does your wireless adaptor have ?

DG1FI, 2024-05-20:
the commands are working. with "bluetoothctl scan on" it found many bluetooth devices, like mobile phones and bluetooth speatker. the blueducky script found only my lge tv. i've tried it many times and with different adapters... 

without the commands above, the adapters are not working -> no compatible adapters found

DG1FI, 2024-05-20:
`  # List devices to see if any are connected
    result = subprocess.run(['bluetoothctl', 'devices'], capture_output=True, text=True)
    if "Device" not in result.stdout:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.")
        return False`

i've try it out... the command "bluetoothctl devices" is here empty at startup... the command "scan on" initiates the scan and add devices to the list, like this:

`
Device 48:E1:5C:76:34:B2 48-E1-5C-76-34-B2
Device 3E:55:13:F2:A9:72 3E-55-13-F2-A9-72
Device 74:FF:8F:4C:5E:76 74-FF-8F-4C-5E-76
Device 61:76:CB:39:FD:7F 61-76-CB-39-FD-7F
Device 18:58:80:51:30:89 [LG] webOS TV UQ70006LB
Device C4:35:34:37:12:75 Govee_H6072_1275
Device C4:42:25:D7:3B:34 T-Echo DG1FI-11
[bluetooth]#
`

but if i run blueducky, only my lge tv is listed- the other devices not.

Aggelos11, 2024-05-21:
Same for me …

Flynnlin, 2024-05-22:
> 在带有 ubuntu 服务器的树莓派 5 上，我解决了这个问题：
> 
> 重新启动蓝牙 `sudo systemctl restart bluetooth`
> 
> 启动进程： `bluetoothctl`
> 
> 执行命令：
> 
> ```
> power on
> agent on
> default-agent
> scan on
> ```
> 
> 退出bluetoothctl `exit`
> 
> 之后： `sudo python3 BlueDucky.py`将开始。

worked

Aggelos11, 2024-05-22:
It worked for me once but I need to repeat this process each time I want to run the script . Also most times it does not work ( does not find Bluetooth devices )

DG1FI, 2024-05-23:
try it on a another laptop same issue, only the tv's was found, but bluetoothctl found a lot of devices

[image]

[image]

Christiancoding, 2024-05-24:
I had the same issue:
{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.

I fixed it by connecting it to a Bluetooth device. After that, it worked just fine.

Aggelos11, 2024-05-24:
And how exactly did you do that ?

mohamedmajid91, 2024-05-24:
#!/bin/bash
Script to unblock and restore Bluetooth after a crash of BlueDucky.py

echo "Restoring Bluetooth..."
Unblock the Bluetooth device

if ! sudo rfkill unblock bluetooth; then
echo "Failed to unblock Bluetooth device."
exit 1
fi
Bring up the Bluetooth interface

if ! sudo hciconfig hci1 up; then
echo "Failed to bring up Bluetooth interface."
exit 1
fi
Restart the Bluetooth service

if ! sudo systemctl restart bluetooth; then
echo "Failed to restart Bluetooth service."
exit 1
fi
Wait for the service to restart

sleep 5
Use bluetoothctl to configure Bluetooth

{
echo "power on"
sleep 1
echo "agent off"
sleep 1
echo "agent on"
sleep 1
echo "default-agent"
sleep 1
echo "scan on"
sleep 5
} | bluetoothctl

echo "Bluetooth restored successfully."
Delay before running the Python script

sleep 5
Run the Python script

if ! sudo python3 BlueDucky.py; then
echo "Failed to run BlueDucky.py."
exit 1
fi

exit 0

sudo nano restore_bluetooth.sh

sudo chmod +x restore_bluetooth.sh

ABC00012345, 2024-06-07:
or just remove the return False in the script, it is working for me

Aggelos11, 2024-06-07:
Can you explain it a bit more please ? Exactly what I need to do in which line etc?

> or just remove the return False in the script, it is working for me

ABC00012345, 2024-06-07:
> Can you explain it a bit more please ? Exactly what I need to do in which line etc?
> 
> > or just remove the return False in the script, it is working for me

Here, I think it's near to line 640, the last return False, where it says before no compltatible bluetooth devices connected. Or you can also remove the complete if condition: 

# Check if bluetoothctl is available
    try:
        subprocess.run(['bluetoothctl', '--version'], check=True, stdout=subprocess.PIPE)
    except subprocess.CalledProcessError:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: {blue}bluetoothctl {reset}is not installed or not working properly.")
        return False

    # Check for Bluetooth adapters
    result = subprocess.run(['bluetoothctl', 'list'], capture_output=True, text=True)
    if "Controller" not in result.stdout:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: No {blue}Bluetooth adapters{reset} have been detected.")
        return False

    # List devices to see if any are connected
    result = subprocess.run(['bluetoothctl', 'devices'], capture_output=True, text=True)
    if "Device" not in result.stdout:
        print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.")
        return False # HERE ....................... Remove this

    # if no issues are found then continue
    return True

Robotic31, 2024-07-06:
> Pada raspberry Pi 5 dengan server ubuntu saya selesaikan dengan ini:
> 
> Mulai ulang bluetoothnya `sudo systemctl restart bluetooth`
> 
> Mulai prosesnya: `bluetoothctl`
> 
> jalankan perintah:
> 
> ```
> power on
> agent on
> default-agent
> scan on
> ```
> 
> Keluar dari bluetoothctl `exit`
> 
> setelah itu: `sudo python3 BlueDucky.py`akan dimulai.

good its work

Kami-Sama123, 2024-07-10:
> On raspberry Pi 5 with ubuntu server i solved with this:
> 
> Restart the bluetooth `sudo systemctl restart bluetooth`
> 
> Start the process: `bluetoothctl`
> 
> execute command:
> 
> ```
> power on
> agent on
> default-agent
> scan on
> ```
> 
> Exit from bluetoothctl `exit`
> 
> after that: `sudo python3 BlueDucky.py` will start.

Bro Alex you are amazing

harshads-dev, 2024-09-09:
can i run this on my laptop?

harshads-dev, 2024-09-09:
@alex8819 @mohamedmajid91 @Christiancoding @Flynnlin  @Aggelos11 will this code run on my laptop?

Jaron-Wilson, 2024-09-15:
> After BlueDucky.py crash, every time i need to make this command 2 times, after that BlueDucky.py start, but when it will crash i need to do it again. i mada a small script to speed up: `sudo nano [...]

Yours dont work so i made one:
#!/bin/bash

# Restart Bluetooth service
echo "Restarting Bluetooth service..."
sudo systemctl restart bluetooth

# Start Bluetoothctl and execute commands
echo "Starting bluetoothctl and executing commands..."
(
    echo "power on"
    sleep 1
    echo "agent on"
    sleep 1
    echo "default-agent"
    sleep 1
    echo "scan on"
    sleep 5  # Adjust the delay as needed for scanning
    echo "exit"
) | bluetoothctl

# Start BlueDucky.py
echo "Starting BlueDucky.py..."
sudo python3 BlueDucky.py

MohammadParsaTheFirst, 2024-09-16:
thank you for your guidance --- it worked for me

harshads-dev, 2024-09-16:
@Jaron-Wilson brother i get error port 17 refused error even in android 11 and I tried on android 9 its sends a pair passkey for pairing it doesn't work, any thing to fix? Or which bluez version i should install in my laptop?

Jaron-Wilson, 2024-09-16:
The port 17 is that android and iso forced an update without permission, when you turn on your phone and don't restart it it does the update in background, I used ble low power mode and that's the one for watches and that works. Error port 17 is patched, and the people who made this only posted the code after it got patched.

harshads-dev, 2024-09-21:
@Jaron-Wilson Can you tell how you setup blueducky and bluez for this? which version of bluez you installed, when i tried on older device i get "ERROR - [TX-19] Bluetooth error: [Errno 104] Connection reset by peer"
can you tell me how you setup this tool, bluez and blueducky, pybluez these

thank you for support brother

victordonat0, 2024-09-27:
> The port 17 is that android and ios forced an update without permission, when you turn on your phone and don't restart it it does the update in background, I used ble low power mode and that's the one for watches and that works. Error port 17 is patched, and the people who made this only posted the code after it got patched.

working with my orange pi 3b i test on a A10

aayush022008, 2025-05-06:
same problem not solve in mine

Whoami-Parrot, 2025-06-27:
Hey,

The error is in the code.

Go check my repository, I have probably fixed this:
https://github.com/Whoami-Parrot/BlueDucky

If you see two BlueDucky files. Please run the BlueDucky.py, the other was added by me by mistake

Whoami-Parrot, 2025-06-27:
It's a code issue in line ~663  (a return false causes this) 

To you and people who are here. Check my repository, I have fixed this:
https://github.com/Whoami-Parrot/BlueDucky.git

If it doesn't work, let me know

Whoami-Parrot, 2025-06-27:
Hey, this is a issue in the code at kine ~663. 
A "return false" causes this

I have fixed it, check this repository:
https://github.com/Whoami-Parrot/BlueDucky.git