pentestfunctions/BlueDucky #51: Not completing script and getting stuck

Hypoonsil, 2024-05-20:
So i was following the tutorial on the readme.md while in the last stage the script doesn't run fully....
Here is a screenshot:
[image]

madarauchia7387438, 2024-05-21:
yes you are right ...i am also figuring this issue ...please help me

Roshan-777, 2024-05-21:
Same issue, cant figure out about what to do to resolve this.

Bigyan-009, 2024-05-21:
same isuue please anyone figure it out

coldvisionz, 2024-05-21:
fixed this by enabling bluetooth on kali via systemd ctl, because bluetooth daemon is disabled by default for security reasons
$ `systemctl enable bluetooth`
$ `systemctl start bluetooth`

Bigyan-009, 2024-05-22:
> fixed this by enabling bluetooth on kali via systemd ctl, because bluetooth daemon is disabled by default for security reasons $ `systemctl enable bluetooth` $ `systemctl start bluetooth`

[image]
after doing this i am again facing this issue

pentestfunctions, 2024-05-22:
Are you doing this on a raspberry pi as stated in the readme?

Bigyan-009, 2024-05-23:
no i am doing it in kali linux in my laptop with internal bluetooth hardware

mohamedmajid91, 2024-05-24:
#!/bin/bash
Script to unblock and restore Bluetooth after a crash of BlueDucky.py

echo "Restoring Bluetooth..."
Unblock the Bluetooth device

if ! sudo rfkill unblock bluetooth; then
echo "Failed to unblock Bluetooth device."
exit 1
fi
Bring up the Bluetooth interface

if ! sudo hciconfig hci1 up; then
echo "Failed to bring up Bluetooth interface."
exit 1
fi
Restart the Bluetooth service

if ! sudo systemctl restart bluetooth; then
echo "Failed to restart Bluetooth service."
exit 1
fi
Wait for the service to restart

sleep 5
Use bluetoothctl to configure Bluetooth

{
echo "power on"
sleep 1
echo "agent off"
sleep 1
echo "agent on"
sleep 1
echo "default-agent"
sleep 1
echo "scan on"
sleep 5
} | bluetoothctl

echo "Bluetooth restored successfully."
Delay before running the Python script

sleep 5
Run the Python script

if ! sudo python3 BlueDucky.py; then
echo "Failed to run BlueDucky.py."
exit 1
fi

exit 0

sudo nano restore_bluetooth.sh

sudo chmod +x restore_bluetooth.sh

Enveeus, 2024-05-25:
worked for me, thanks!

Hypoonsil, 2024-05-25:
> fixed this by enabling bluetooth on kali via systemd ctl, because bluetooth daemon is disabled by default for security reasons
> $ `systemctl enable bluetooth`
> $ `systemctl start bluetooth`
> 

Well thanks it worked

AzhariRamadhan, 2024-06-07:
```
#!/bin/bash
# Script to unblock and restore Bluetooth after a crash of BlueDucky.py

echo "Restoring Bluetooth..."

# Unblock the Bluetooth device
if ! sudo rfkill unblock bluetooth; then
    echo "Failed to unblock Bluetooth device."
    exit 1
fi

# Bring up the Bluetooth interface (try hci0)
if ! sudo hciconfig hci0 up; then
    echo "Failed to bring up Bluetooth interface."
    exit 1
fi

# Restart the Bluetooth service
if ! sudo systemctl restart bluetooth; then
    echo "Failed to restart Bluetooth service."
    exit 1
fi

# Wait for the service to restart
sleep 5

# Use bluetoothctl to configure Bluetooth
{
    echo "power on"
    sleep 1
    echo "agent off"
    sleep 1
    echo "agent on"
    sleep 1
    echo "default-agent"
    sleep 1
    echo "scan on"
    sleep 5
} | bluetoothctl

echo "Bluetooth restored successfully."

# Delay before running the Python script
sleep 5

# Run the Python script
if ! sudo python3 BlueDucky.py; then
    echo "Failed to run BlueDucky.py."
    exit 1
fi

exit 0

```

harshads-dev, 2024-09-09:
> no i am doing it in kali linux in my laptop with internal bluetooth hardware
@Bigyan-009 How You are Doing? I can't Run I Get this error 2024-09-09 19:31:30,980 - ERROR - ERROR connecting on port 19: [Errno 104] Connection reset by peer
2024-09-09 19:31:30,981 - ERROR - Connection failure: Connection failure on port 19
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (104, 'Connection reset by peer')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 265, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 104] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 697, in <module>
    main()
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 679, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 626, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 618, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 158, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 158, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/harshad/Downloads/BlueDucky-49f470aa979d56b2ad5911b54ed38c39d28aa17c/BlueDucky.py", line 273, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 19

@pentestfunctions help me using inbuilt Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad] in my laptop

chrisytharp, 2025-01-12:
> ```
> #!/bin/bash
> # Script to unblock and restore Bluetooth after a crash of BlueDucky.py
>
> echo "Restoring Bluetooth..."
>
> # Unblock the Bluetooth device
> if ! sudo rfkill unblock bluetooth; then [...]

This worked perfectly!!! I had to change the scipt to "hci0" since in using the onboard bluetooth! (raspberry pi 4 x64 with kali x64)