pentestfunctions/BlueDucky #52: How can I solve this?(ERROR-Unable to set adapter name)

Vamerio, 2024-05-22:
After I choose the payload, It shows this 
[image]

Aggelos11, 2024-05-24:
Hey which Bluetooth adaptor do you use ?

mohamedmajid91, 2024-05-25:
> Hey which Bluetooth adaptor do you use ?

ud100

Vamerio, 2024-06-07:
> > Hey which Bluetooth adaptor do you use ?
> 
> 
> 
> ud100
> 
> 

Sorry for the late response,the name of the adaptor is Cambridge Silicon Radio Ltd.

Aggelos11, 2024-06-07:
Can you send us an Amazon link to this adaptor ?

moadpy, 2024-09-22:
[image]

Delete line 615 in the code. That line tries to change the name of the adapters, which causes the error. After deleting it, the script worked for me