pentestfunctions/BlueDucky #53: how to solve ERROR - ERROR connecting on port 17: [Errno 111]

Sujalcs, 2024-05-22:
i am on a laptop using kali i have install all the requirements i tried to use this on samsung A2 core on the first attempt it opened google and then the script suddenly stopped and now i am getting this error 

Available payloads:                                                                         
[1]: payload_example_2.txt
[2]: payload_example_1.txt
[3]: wp_payload.txt

Enter the number that represents the payload you would like to load: 1
Selected payload: /home/sujal/BlueDucky/payloads/payload_example_2.txt
2024-05-23 19:43:29,390 - INFO - executing 'sudo service bluetooth restart'
[sudo] password for sujal: 
2024-05-23 19:43:34,041 - INFO - executing 'sudo hciconfig hci0 name Robot POC'
2024-05-23 19:43:34,096 - INFO - executing 'hciconfig hci0 name'
2024-05-23 19:43:34,114 - INFO - executing 'sudo hciconfig hci0 class 9536'
2024-05-23 19:43:34,134 - INFO - executing 'hciconfig hci0 class'
2024-05-23 19:43:34,140 - INFO - executing 'sudo hciconfig hci0 sspmode 1'
2024-05-23 19:43:36,412 - INFO - connecting to 38:EA:C1:3B:AE:AB on port 1
2024-05-23 19:43:36,854 - INFO - connecting to 38:EA:C1:3B:AE:AB on port 17
2024-05-23 19:43:37,010 - ERROR - ERROR connecting on port 17: [Errno 111] Connection refused                                                                                           
2024-05-23 19:43:37,010 - ERROR - Connection failure: Connection failure on port 17
-e remove 38:EA:C1:3B:AE:ABoothd...[bluetooth]#         Leaf Bass 2
Invalid command in menu main: -e

Use "help" for a list of available commands in a menu.
Use "menu <submenu>" if you want to enter any submenu.
Use "back" if you want to return to menu main.
[Leaf Bass 2]# Successfully Removed device: 38:EA:C1:3B:AE:AB
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (111, 'Connection refused')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 264, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 755, in <module>
    main()
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 728, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 637, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 629, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 157, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 157, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/sujal/BlueDucky/./BlueDucky.py", line 278, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 17

Aggelos11, 2024-05-24:
Hey which Bluetooth adaptor do you use ?

Enveeus, 2024-05-25:
So I ran into this issue myself and was stuck for a long time trying to resolve. This is not an issue, what this means is that the device you are attacking is not susceptible to the exploit and the blueducky script terminates the connection.  when you find a device that is susceptible the script will complete.

Sujalcs, 2024-05-26:
> Hey which Bluetooth adaptor do you use ?
i am using Qualcomm QCA9377

Sujalcs, 2024-05-26:
> So I ran into this issue myself and was stuck for a long time trying to resolve. This is not an issue, what this means is that the device you are attacking is not susceptible to the exploit and the [...]

the device i was tragetting was using android 8 thats why i was not sure if its a bug or something else is wrong

Enveeus, 2024-05-26:
> > Hey which Bluetooth adaptor do you use ?
> > i am using Qualcomm QCA9377

I'm using the Panda 4.0., (Link Below). It's a Cambridge Silicon Radio adapter and when you lsusbe it states  Cambridge Silicon Radio (HCI Mode)

Amazon Link:
https://amzn.to/3Ki6l6s

Aggelos11, 2024-05-26:
Wait so you use the panda adaptor and you get error ? Which exact error? Please show me

Enveeus, 2024-05-26:
What you see in the first post above is the error you will get **_WHEN_** the device you are attacking is not susceptible to this exploit or has been patched, even with a working adapter such as the Panda. 

The script is telling you right here (below) )that the other device is refusing your connection request:

2024-05-23 19:43:37,010 - ERROR - ERROR connecting on port 17: [Errno 111] **Connection refused**

I've run this on a few Android devices including a Galaxy S10 (Adroind 12), Nvidia Shield, and Android A10 (Android 9). The script only completes on the A10 but **WILL NOT** run the exploit of opening up a webpage to YouTube.

Aggelos11, 2024-05-26:
But is the panda adaptor working fine and out of the box ? Or do you generally fake any issues when using it ?

Enveeus, 2024-05-26:
Out the box it works, no issues here, so yes Plug-n-Play

Aggelos11, 2024-05-26:
Do you need to adjust any settings with Bluetoothctl power on /agent on etc? Or you just plug in the adaptor and run the script via Python3 BlueDucky.py?

Sujalcs, 2024-05-26:
> What you see in the first post above is the error you will get **_WHEN_** the device you are attacking is not susceptible to this exploit or has been patched, even with a working adapter such as the [...]

Alright then I will try to target some other device let's hope it works

Aggelos11, 2024-05-26:
Let us know how it goes

Dmorr157, 2024-06-05:
did this work on another device?

Shaurya-Grover, 2024-06-08:
Hello everybody so I was running the BlueDucky Script luckily it scanned all my devices but when I wanted to attack one of them I tried using a payload named `payload_example_1.txt` and when I entered the payload and ran it. I saw the below error pop up on my screen :

```
2024-06-08 17:13:12,859 - ERROR - ERROR connecting on port 17: [Errno 103] Software caused connection abort
2024-06-08 17:13:12,860 - ERROR - Connection failure: Connection failure on port 17
Agent registered
[CHG] Controller D8:3A:DD:28:0D:C2 Powered: no
[CHG] Controller D8:3A:DD:28:0D:C2 Discovering: no
[CHG] Controller D8:3A:DD:28:0D:C2 Class: 0x00000000
[bluetooth]# -e remove 70:BB:E9:8F:99:3B
Invalid command in menu main: -e

Use "help" for a list of available commands in a menu.
Use "menu <submenu>" if you want to enter any submenu.
Use "back" if you want to return to menu main.
Successfully Removed device: 70:BB:E9:8F:99:3B
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (103, 'Software caused connection abort')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/raspberrypi/BlueDucky/BlueDucky.py", line 264, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 103] Software caused connection abort
```

which basically ment that it failed to connect on PORT 17, and also a random message popped up on my screen once the process of Blueducky was terminated on my raspberry pi. The message in my mobile was -

 `Couldnt pair with ROBOT POC because of an incorrect pin or passkey`
 
 I was hoping if somebody would help me running the BlueDucky Script, your guidance and assistant would be highly appreciated!
 
Best Regards 
Shaurya Grover

Aggelos11, 2024-07-20:
> Hi, @Shaurya-Grover The problem is caused by the protection of bluetooth in variety of phone provider (some has restriction on keyboard devices but some wont🤔) , or even just the device has patched (Security bulletin 2023-12)
> 
> But you can solve the problem by adding a loop , it will retry until it succed ! (not include patched devices)

Can you send the full code pls ?

Shaurya-Grover, 2024-07-20:
@Heyholiday067
Do you remember which function you specifically modified and added a loop in, so that it kept on retrying until it got the access?

If yes kindly share it would be of great help!

Regards
Shaurya