pentestfunctions/BlueDucky #65: error after running script

shivam9312, 2024-05-30:
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (112, 'Host is down')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/vilen/BlueDucky/BlueDucky.py", line 264, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 112] Host is down

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/vilen/BlueDucky/BlueDucky.py", line 756, in <module>
    main()
  File "/home/vilen/BlueDucky/BlueDucky.py", line 729, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vilen/BlueDucky/BlueDucky.py", line 637, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/vilen/BlueDucky/BlueDucky.py", line 629, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vilen/BlueDucky/BlueDucky.py", line 157, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vilen/BlueDucky/BlueDucky.py", line 157, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/vilen/BlueDucky/BlueDucky.py", line 278, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 1

garybolanos1, 2024-09-17:
have you found any workaround? I have the exact same issue

jalupaja, 2024-12-01:
The error is saying that the "Host is down", meaning the device isn't in range or the Bluetooth Address is wrong.