pentestfunctions/BlueDucky #68: Adapter not compatibale!!!

yay-whyawhy, 2024-06-01:
llah㉿windows)-[~/Downloads/BlueDucky]
└─$ ls   
BlueDucky.py  README.md  __init__.py  images  payloads  requirements.txt  reset.sh  resete.sh  utils
                                                                                                                                                                        
┌──(abdullah㉿windows)-[~/Downloads/BlueDucky]
└─$ ./resete.sh         
Restoring Bluetooth...
Waiting to connect to bluetoothd...[bluetooth]# power on
[bluetooth]# Agent registered
[bluetooth]# Changing power on succeeded
[bluetooth]# agent off
[bluetooth]# Agent unregistered
[bluetooth]# agent on
[bluetooth]# Agent registered
[bluetooth]# default-agent
[bluetooth]# Default agent request successful
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Bluetooth restored successfully.
========================================================================================================================================================================
                                                                 BlueDucky - Bluetooth Device Attacker                                                                  
                                                                                Ver 2.1                                                                                 
========================================================================================================================================================================
Remember, you can still attack devices without visibility..
If you have their MAC address..
========================================================================================================================================================================
{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.
                                                                                                                                                                        
┌──(abdullah㉿windows)-[~/Downloads/BlueDucky]
└─$ ./reset.sh 
Failed to set power on: org.bluez.Error.Busy
[CHG] Controller 00:1A:7D:DA:71:13 Discoverable: yes
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001104-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00005005-0000-1000-8000-0002ee000001
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001106-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001104-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00005005-0000-1000-8000-0002ee000001
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001106-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:1A:7D:DA:71:13 Class: 0x00100000 (1048576)
[CHG] Controller 00:1A:7D:DA:71:13 UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[... 48 lines cut ...]
	RX bytes:497 acl:0 sco:0 events:21 errors:0
	TX bytes:332 acl:0 sco:0 commands:21 errors:0

                                                                                                                                                                        
┌──(abdullah㉿windows)-[~/Downloads/BlueDucky]
└─$ hcitool scan
Scanning ...
	5C:C5:63:1B:49:BE	Android TV
	2C:4D:DE:04:AB:D4	TECNO POP 5 LTE
                                                                                                                                                                        
┌──(abdullah㉿windows)-[~/Downloads/BlueDucky]
└─$ sudo python3 BlueDucky.py        
========================================================================================================================================================================
                                                                 BlueDucky - Bluetooth Device Attacker                                                                  
                                                                                Ver 2.1                                                                                 
========================================================================================================================================================================
Remember, you can still attack devices without visibility..
If you have their MAC address..
========================================================================================================================================================================
{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.
                                                                                                                                                                        
┌──(abdullah㉿windows)-[~/Downloads/BlueDucky]
└─$ 

Plzzzzz help me resolve the issue. I've tried almost everything, but nothing seems to work. I'm open to any other solutions.

Dovis12341, 2024-06-09:
buy yourself a raspberry pi 4

jalupaja, 2024-12-01:
I used an ASUS BT-500 which is much cheaper then a raspi.