pentestfunctions/BlueDucky #69: No {blue}Bluetooth adapters{reset} have been detected

BG7UHS, 2024-06-02:
The previous version is still usable

[image]

Neowizard, 2024-06-06:
It happens when `bluetoothctl devices` returns an empty list. It's a strange test, since you can attack a BT device that isn't broadcasting.

For playing around and POC, just remove the check on lines 661-665
```
    # List devices to see if any are connected
    result = subprocess.run(['bluetoothctl', 'devices'], capture_output=True, text=True)
    #if "Device" not in result.stdout:
        #print("{reset}[{red}!{reset}] {red}CRITICAL{reset}: No Compatible {blue}Bluetooth devices{reset} are connected.")
        #return False
```

BG7UHS, 2024-06-06:
extremely grateful

[image]

closed 2024-06-07 by BG7UHS

Kozin-Ko, 2024-12-28:
I am a beginner, Sir. Could you please teach me how to remove this line 661-665 as I have faced the same error like you. Thanks.

HansHustler, 2025-02-19:
i ve the same issue on raspberry pi 4 , i deleted the part you said ,  but still the same error :/

Neowizard, 2025-02-19:
> i ve the same issue on raspberry pi 4 , i deleted the part you said , but still the same error :/

It's highly unlikely that you deleted the `print` statement, and are still getting the same error that includes that print statement. 

I assume the error you're experiencing is not the same, or you're not actually running the code with those lines removed.