pentestfunctions/BlueDucky #72: requerments

dirtyhacke, 2024-06-12:
error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      + meson setup /tmp/pip-install-c053pt46/pygobject_1b7efc7f96af412993fb94d305c27621 /tmp/pip-install-c053pt46/pygobject_1b7efc7f96af412993fb94d305c27621/.mesonpy-iqlyt5yu -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md -Dtests=false -Dwheel=true --wrap-mode=nofallback --native-file=/tmp/pip-install-c053pt46/pygobject_1b7efc7f96af412993fb94d305c27621/.mesonpy-iqlyt5yu/meson-python-native-file.ini
      The Meson build system
      Version: 1.4.1
      Source dir: /tmp/pip-install-c053pt46/pygobject_1b7efc7f96af412993fb94d305c27621
      Build dir: /tmp/pip-install-c053pt46/pygobject_1b7efc7f96af412993fb94d305c27621/.mesonpy-iqlyt5yu
      Build type: native build
      Project name: pygobject
      Project version: 3.48.1
      C compiler for the host machine: cc (gcc 13.2.0 "cc (Debian 13.2.0-23) 13.2.0")
      C linker for the host machine: cc ld.bfd 2.42
      Host machine cpu family: x86_64
      Host machine cpu: x86_64
      Program python3 found: YES (/usr/bin/python3)
      Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
      Run-time dependency python found: YES 3.11
      Found CMake: /usr/bin/cmake (3.28.3)
      Run-time dependency gobject-introspection-1.0 found: NO (tried pkgconfig and cmake)
      Not looking for a fallback subproject for the dependency gobject-introspection-1.0 because:
      Use of fallback dependencies is disabled.
      
      ../meson.build:29:9: ERROR: Dependency 'gobject-introspection-1.0' is required but not found.
      
      A full log can be found at /tmp/pip-install-c053pt46/pygobject_1b7efc7f96af412993fb94d305c27621/.mesonpy-iqlyt5yu/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Sergimedita, 2024-09-09:
I have the same error, did you found a solution?

jalupaja, 2024-11-30:
Have you tried installing all packages via apt or pacman and not using the pip packages?
This worked for me.

dirtyhacke, 2024-12-27:
@Sergimedita noh  brohh!!

dirtyhacke, 2024-12-27:
@Sergimedita  did  you find it??

dirtyhacke, 2024-12-27:
@jalupaja  thanks brohh!!