pentestfunctions/BlueDucky #75: Blue ducky

31vky, 2024-06-22:


mohamedmajid91, 2024-06-24:
Known devices:
1: Device Name: Infinix NOTE 40, Address: 7C:8B:C1:01:B3:74

Do you want to use one of these known devices? (yes/no): yes
Enter the index number of the device to attack: 1

 Would you like to register this device:
Infinix NOTE 40 7C:8B:C1:01:B3:74? (y/n) y

Available payloads:
[1]: payload_example_1.txt
[2]: wp_payload.txt
[3]: payload_example_2.txt

Enter the number that represents the payload you would like to load: 1
Selected payload: /home/kali/BlueDucky/payloads/payload_example_1.txt
2024-06-24 22:07:07,278 - INFO - executing 'sudo service bluetooth restart'
2024-06-24 22:07:07,925 - INFO - executing 'sudo hciconfig hci1 name Robot POC'
2024-06-24 22:07:07,941 - INFO - executing 'hciconfig hci1 name'
2024-06-24 22:07:07,960 - INFO - executing 'sudo hciconfig hci1 class 9536'
2024-06-24 22:07:07,975 - INFO - executing 'hciconfig hci1 class'
2024-06-24 22:07:07,979 - INFO - executing 'sudo hciconfig hci1 sspmode 1'
2024-06-24 22:07:10,248 - INFO - connecting to 7C:8B:C1:01:B3:74 on port 1
2024-06-24 22:07:11,532 - INFO - connecting to 7C:8B:C1:01:B3:74 on port 17
2024-06-24 22:07:12,074 - ERROR - ERROR connecting on port 17: [Errno 111] Connection refused
2024-06-24 22:07:12,075 - ERROR - Connection failure: Connection failure on port 17
-e remove 7C:8B:C1:01:B3:74oothd...[bluetooth]#         Infinix NOTE 40
Invalid command in menu main: -e

Use "help" for a list of available commands in a menu.
Use "menu <submenu>" if you want to enter any submenu.
Use "back" if you want to return to menu main.
[Infinix NOTE 40]# hci1 new_settings: powered bondable ssp br/edr le secure-conn 
[Infinix NOTE 40]# hci0 new_settings: powered bondable ssp br/edr le secure-conn 
[Infinix NOTE 40]# Successfully Removed device: 7C:8B:C1:01:B3:74
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (111, 'Connection refused')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line 264, in connect
    sock.connect((self.addr, self.port))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kali/BlueDucky/BlueDucky.py", line 755, in <module>
    main()
  File "/home/kali/BlueDucky/BlueDucky.py", line 728, in main
    hid_interrupt_client = setup_and_connect(connection_manager, target_address, adapter_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 637, in setup_and_connect
    establish_connections(connection_manager)
  File "/home/kali/BlueDucky/BlueDucky.py", line 629, in establish_connections
    if not connection_manager.connect_all():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 157, in connect_all
    return sum(client.connect() for client in self.clients.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 157, in <genexpr>
    return sum(client.connect() for client in self.clients.values())
               ^^^^^^^^^^^^^^^^
  File "/home/kali/BlueDucky/BlueDucky.py", line 278, in connect
    raise ConnectionFailureException(f"Connection failure on port {self.port}")
ConnectionFailureException: Connection failure on port 17
Failed to run BlueDucky.py.

jalupaja, 2024-11-30:
The Infinix NOTE 40 was released in 2024.
All devices updated after December 2023 are fixed and won't work.