pentestfunctions/BlueDucky #78: Failed install dbus

dalpan, 2024-07-14:
How to fix this ?

```
Collecting dbus-python==1.3.2 (from -r requirements.txt (line 1))
  Using cached dbus-python-1.3.2.tar.gz (605 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [95 lines of output]
      + meson setup /tmp/pip-install-px8dx3m5/dbus-python_18b28a7898874f29b44b102e8b428ac8 /tmp/pip-install-px8dx3m5/dbus-python_18b28a7898874f29b44b102e8b428ac8/.mesonpy-o_chfx0n -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-px8dx3m5/dbus-python_18b28a7898874f29b44b102e8b428ac8/.mesonpy-o_chfx0n/meson-python-native-file.ini
      The Meson build system
      Version: 1.5.0
      Source dir: /tmp/pip-install-px8dx3m5/dbus-python_18b28a7898874f29b44b102e8b428ac8
      Build dir: /tmp/pip-install-px8dx3m5/dbus-python_18b28a7898874f29b44b102e8b428ac8/.mesonpy-o_chfx0n
      Build type: native build
      Project name: dbus-python
      Project version: 1.3.2
      C compiler for the host machine: cc (gcc 13.2.0 "cc (Debian 13.2.0-25) 13.2.0")
      C linker for the host machine: cc ld.bfd 2.42.50.20240625
      Host machine cpu family: x86_64
      Host machine cpu: x86_64
      Compiler for C supports arguments -fno-common: YES
      Compiler for C supports arguments -Wno-missing-field-initializers: YES
      Compiler for C supports arguments -Wno-declaration-after-statement: YES
      Compiler for C supports arguments -Wno-inline: YES
      Compiler for C supports arguments -Wno-redundant-decls: YES
      Compiler for C supports arguments -Wno-switch-default: YES
      Compiler for C supports arguments -Wno-write-strings: YES
      Compiler for C supports arguments -Wcast-align: YES
      Compiler for C supports arguments -Wdouble-promotion: YES
      Compiler for C supports arguments -Wduplicated-cond: YES
      Compiler for C supports arguments -Wfloat-equal: YES
      Compiler for C supports arguments -Wformat-nonliteral: YES
      Compiler for C supports arguments -Wformat-security: YES
      Compiler for C supports arguments -Wformat=2: YES
      Compiler for C supports arguments -Winit-self: YES
      Compiler for C supports arguments -Wlogical-op: YES
      Compiler for C supports arguments -Wmissing-declarations: YES
      Compiler for C supports arguments -Wmissing-format-attribute: YES
      Compiler for C supports arguments -Wmissing-include-dirs: YES
      Compiler for C supports arguments -Wmissing-noreturn: YES
      Compiler for C supports arguments -Wnull-dereference: YES
      Compiler for C supports arguments -Wpacked: YES
      Compiler for C supports arguments -Wpointer-arith: YES
      Compiler for C supports arguments -Wshadow: YES
      Compiler for C supports arguments -Wswitch-enum: YES
      Compiler for C supports arguments -Wundef: YES
      Compiler for C supports arguments -Wunused-but-set-variable: YES
      Compiler for C supports arguments -Wjump-misses-init: YES
      Compiler for C supports arguments -Wmissing-prototypes: YES
      Compiler for C supports arguments -Wnested-externs: YES
      Compiler for C supports arguments -Wold-style-definition: YES
      Compiler for C supports arguments -Wpointer-sign: YES
      Compiler for C supports arguments -Wstrict-prototypes: YES
      Configuring _dbus-python-config.h using configuration

      Executing subproject dbus-gmain

      dbus-gmain| Project name: dbus-gmain
      dbus-gmain| Project version: undefined
      dbus-gmain| C compiler for the host machine: cc (gcc 13.2.0 "cc (Debian 13.2.0-25) 13.2.0")
      dbus-gmain| C linker for the host machine: cc ld.bfd 2.42.50.20240625
      dbus-gmain| Compiler for C supports arguments -fno-common: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wno-missing-field-initializers: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wcast-align: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wdouble-promotion: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wduplicated-branches: YES
      dbus-gmain| Compiler for C supports arguments -Wduplicated-cond: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wfloat-equal: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wformat-nonliteral: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wformat-security: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wformat=2: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Winit-self: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wlogical-op: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wmissing-declarations: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wmissing-format-attribute: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wmissing-include-dirs: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wmissing-noreturn: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wnull-dereference: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wpacked: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wpointer-arith: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wredundant-decls: YES
      dbus-gmain| Compiler for C supports arguments -Wshadow: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wswitch-default: YES
      dbus-gmain| Compiler for C supports arguments -Wswitch-enum: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wundef: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wunused-but-set-variable: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wwrite-strings: YES
      dbus-gmain| Compiler for C supports arguments -Wdeclaration-after-statement: YES
      dbus-gmain| Compiler for C supports arguments -Wjump-misses-init: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wmissing-prototypes: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wnested-externs: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wold-style-definition: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wpointer-sign: YES (cached)
      dbus-gmain| Compiler for C supports arguments -Wstrict-prototypes: YES (cached)
      dbus-gmain| Did not find pkg-config by name 'pkg-config'
      dbus-gmain| Found pkg-config: NO
      dbus-gmain| Did not find CMake 'cmake'
      dbus-gmain| Found CMake: NO
      dbus-gmain| Run-time dependency dbus-1 found: NO

      ../subprojects/dbus-gmain/meson.build:107:11: ERROR: Dependency lookup for dbus-1 with method 'pkgconfig' failed: Pkg-config for machine host machine not found. Giving up.

      A full log can be found at /tmp/pip-install-px8dx3m5/dbus-python_18b28a7898874f29b44b102e8b428ac8/.mesonpy-o_chfx0n/meson-logs/meson-log.txt
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

Heyholiday067, 2024-07-14:
../subprojects/dbus-gmain/meson.build:107:11: ERROR: Dependency lookup for dbus-1 with method 'pkgconfig' failed: Pkg-config for machine host machine not found. Giving up.

 It told you the problem is here.

4p3rtur3, 2024-07-25:
Googled 2sec: https://stackoverflow.com/questions/61285415/no-package-dbus-1-found
```
sudo apt install build-essential libdbus-glib-1-dev libgirepository1.0-dev libpython3-dev libdbus-1-dev
pip install dbus-python
```
Worked for me.