pentestfunctions/BlueDucky #79: sudo hciconfig hci0 up failed

tower-sun, 2024-07-15:
when [sudo hciconfig hci0 up] it occurs [can't open hci socket.:permission denied] , I've [chmod 777 BlueDucky]
[image]

coldvisionz, 2024-07-16:
nowadays android  kernels block any extended features with radios like wi-fi & bluetooth.
kali nethunter works well with other android vendor models and nexus phone, i recommend using external adapter as @Heyholiday067 mentioned

tower-sun, 2024-07-23:
[image]
why can't I run BlueDucky.py