pentestfunctions/BlueDucky #82: Cannot get to work on Pi Zero W

Shadowdrums, 2024-07-26:
I tried this on my laptop which after reading the readme.md realized it was for raspberry pi. I have  Pi 4 I can try this on but so far, I've had no luck with the pi zero w. I would love to see this working on a Pi Zero W for various use cases, :)

thank you 
Shadowdrums

FaganAfandiyev, 2024-10-26:
Hey @Shadowdrums, Did you resolve this?

Shadowdrums, 2024-10-26:
Thank you for reaching out, I am sad to say I have not resolved the issue. I havnt had a lot of time to tinker with it and have since been busy with other projects. For that I apologize. I can look into it further and report any issues u may still be having it resolves I can find.

DezusAZ, 2025-02-02:
Tried getting it to work on a pi zero 2w and it unfortunately cannot find the Bluetooth adapter. I tried setting hci1 up and the adapter can scan for and identify other Bluetooth devices using the OS Bluetooth software but not using the blue ducky script. :/