pentestfunctions/BlueDucky #90: dose not run

Sergimedita, 2024-09-09:
when I run python 3 Blueducky.py  it just stays lie this:
[image]
 and does not run, also I have a porblem with the requierements.txt , it gives me the following error:
[image]
[image]

wangweihuayaochenggong, 2024-09-14:
我也遇到了这种情况，应该怎么解决请问

garybolanos1, 2024-09-16:
same issue. I'm using a raspberry pi 3, and I've also tried in a VM with same results, it does not work

jalupaja, 2024-11-29:
I had the same issue when installing via pip. 
It worked for me using the packages from apt.