pentestfunctions/BlueDucky #98: problem with pairing

Rex-pl, 2025-03-08:
I have a phone with Android 10. When I run the script, it detects the phone and I can select it. After selecting the payload, it connects to the phone, but then I have to click on the phone to confirm pairing and the script stops. When I first pair the phone with the computer and then run the script, everything works normally.

ex0xe, 2025-03-11:
I had the same problem when using kali as the "attacker" distro, but worked great when using ubuntu as attacker distro. Also, which attacker device are you using? Raspberry PI ?

harshads-dev, 2025-04-05:
> I had the same problem when using kali as the "attacker" distro, but worked great when using ubuntu as attacker distro. Also, which attacker device are you using? Raspberry PI ?

I Use Kali as the "attacker" distro in my laptop in Vmware but still facing same problem